JdeRobot/RoboticsInfrastructure #15: Issue Running F1 on ROS-Noetic

AbsorbedInThought, 2020-08-02:
I am having an issue running the F1 model for ROS-Noetic. Whenever I run it, I can only see an empty world in Gazebo. I have verified the directories and all the models and worlds which are required are present in their correct paths. I get no errors and this is the following output:

```
... logging to /home/taha/.ros/log/a94ad440-d4bd-11ea-9689-4d09133a937f/roslaunch-taha-16864.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://taha:46179/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: noetic
 * /rosversion: 1.15.7
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)

auto-starting new master
process[master]: started with pid [16872]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a94ad440-d4bd-11ea-9689-4d09133a937f
process[rosout-1]: started with pid [16882]
started core service [/rosout]
process[gazebo-2]: started with pid [16885]
process[gazebo_gui-3]: started with pid [16893]
[ INFO] [1596372165.612913416]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1596372165.614514275]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1596372165.737760171]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1596372165.739398514]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[ INFO] [1596372166.030785123]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1596372166.075506168, 0.018000000]: Physics dynamic reconfigure ready.
```

I would appreciate if you could share some steps on how you got this working with ROS-Noetic, so I can check if I am missing something.

CMake: 3.16.3
ROS: Noetic
Gazebo: 11
Ubuntu: 20.04 LTS

dcharrezt, 2020-08-03:
Hi @AbsorbedInThought 
which steps did you followed? did you installed it with make?

AbsorbedInThought, 2020-08-05:
Yes, I installed it with make as well as catkin_make.
The directory is as follow: catkin_workspace/src/f1

AbsorbedInThought, 2020-08-14:
With some changes, I was able to run F1 successfully. These were the following steps:

Installing JdeRobot assets:

git clone https://github.com/JdeRobot/assets
cd assets/jderobot_assets && sudo mkdir build && sudo cd build
sudo cmake .. && sudo make && sudo make install
echo "source /opt/jderobot/share/jderobot/gazebo/assets-setup.sh" >> ~/.bashrc
source ~/.bashrc

Installing the F1 model:

git clone -b noetic-devel https://github.com/JdeRobot/CustomRobots
cd CustomRobots/f1 && sudo mkdir build && sudo cd build
sudo cmake .. -DCMAKE_PREFIX_PATH=/opt/ros/noetic
sudo make && sudo make install

Launching F1:

source /home/USERNAME/CustomRobots/f1/build/devel/setup.bash

cd CustomRobots/f1/launch
roslaunch f1_1_nurburgrinlineROS.launch

jmplaza, 2020-08-15:
Good to hear that. Now you can go on playing with the Formula-1 car and VisualCircuit tool. I reopen this issue so we solve it without installing JdeRobot-assets, which is deprecated. We should copy everything required from JdeRobot-assets into CustomRobots, maybe /opt/jderobot/share/jderobot/gazebo/assets-setup.sh...

closed 2024-11-15 by javizqh