JdeRobot/RoboticsInfrastructure #56: Monaco circuit with a Formula1 car

jmplaza, 2020-11-28:
to be used in RoboticsAcademy exercises on autonomous navigation and in BehaviorStudio. Step by step we are increasing the number of well known circuits of F1 around the world (we already have Nurburgring, Montmeló and Montreal so far :-) )

marck3131, 2023-04-01:
Hello , I've been working on this , since a week now. After trying to make few on Blender , I wasn't getting the results expected, Also because there's not much info regarding the dimensions for the monaco circuit and it's complex turns.
I have this ready on Solidworks right now, and would like to ask where can I find the dimensions of the red_line in existing 3-4 examples of RA/F1, so I can complete by scaling it, adding walls on the side, grass besides that, and the red_line to follow and then export .sdf & meshes.

[image]
es to be imported in Gazebo

jmplaza, 2023-04-01:
Hi @marck3131 ,

the circuit looks very nice! Just put a tentative red line in the middle in the road. It does not need an exact copy of the current Formula1 exercise, just a good approximation. Wide enough to be detected from the onboard camera of the Formula1 car.

After finishing a plane circuit, another interesting idea is to implement also a 3D version of the same circuit. Maybe you can explore [this repo](https://github.com/RoboticsLabURJC/2016-tfg-Alvaro-Villamil). We worked on it some years ago...  For instance: https://github.com/RoboticsLabURJC/2016-tfg-Alvaro-Villamil/tree/master/Mundos_JdeRobot/Monaco and https://github.com/RoboticsLabURJC/2016-tfg-Alvaro-Villamil/tree/master/Monaco

[image]
[image]

jmplaza, 2023-04-01:
Also let's start with a working world, but a low-poly release of it would be better, so Gazebo may run it without consuming much computing power...

marck3131, 2023-04-04:
Hello, those references and the textures look amazing ...
I'll surely have a look on them and also see if I am able to modify them. 
Here's what I tried today, 
[image]
Are we okay with such realistic look ? 
If yes, my concern henceforth would be , if gazebo will allow such colors and such aesthetic. I'll start looking for 'How can we import such world files in gazebo without loosing the imported image'