JdeRobot/RoboticsInfrastructure #70: [Noetic-devel] f1 car is initialised at a wrong angle

Suhas-G, 2021-04-14:
This issue is related to the `noetic-devel` branch. When using this branch in Follow line exercise we get the car initialized at a wrong angle.
[image]

Apart from this, there are also some warnings about materials not being applied and errors about multiple initiations of service on terminal.
[image]

Suhas-G, 2021-04-14:
There are multiple differences between the branches in f1 folder
The files having changes.
[image]

For example, the `f1/meshes/F1.dae` has extra content about the materials.

Would it be fine to use same meshes from melodic-devel branch?

jmplaza, 2021-04-14:
sure @Suhas-G . The best and most up-to-date meshes and robot models are in the melodic-devel branch. Feel free to copy everything you need from it

fqez, 2021-04-16:
Hi @Suhas-G, as far as I know that F1 model is ancient, hence I really don't know the state of it (if it works or it does not). Did you manage to update your models, as @jmplaza stated? Let us know so we can close this issue. Thanks!!

sergiopaniego, 2021-04-16:
PR merged, issue closed!

PR #73 (2021-04-16): update models and meshes from melodic-devel branch 
This PR fixes #70 
I have updated the models and meshes from the `melodic-devel` branch for `f1`, `f1_renault_laser`, `many_circuit` and `simple_circuit` since these had differences in files. I have retained the folders which are extra in `noetic-branch` as it is.
closed 2021-04-16 by sergiopaniego