JdeRobot/RoboticsInfrastructure #192: [noetic] Nested models for f1

pariaspe, 2023-02-15:
Since some models use the same f1 (plus some different plugins), they can be nested to reduce pkg size and increase readability.

hady68, 2023-08-08:
Hey @pariaspe was exploring the repository and came across this issue, I'm a first-timer, can you tell something more about this issue and some steps to solve it, i would like to take it up!

pariaspe, 2023-08-09:
Hi @hady68 , current SDF models have a lot of duplicated code. The idea is to create a base model (f1_base for instance) and include that base model in the final ones. You can take a look at the [SDFormat specification](http://sdformat.org/spec?ver=1.9&elem=sdf) docs.

hady68, 2023-08-09:
> Hi @hady68 , current SDF models have a lot of duplicated code. The idea is to create a base model (f1_base for instance) and include that base model in the final ones. You can take a look at the [SDFormat specification](http://sdformat.org/spec?ver=1.9&elem=sdf) docs. 

I'll take a look

Faizan-Alam-1, 2024-05-03:
Is this issue still open?

jmplaza, 2024-05-03:
Yes @Faizan-Alam-1, it is open. Improving the Gazebo Formula1 car model would be a nice contribution.

Cheers

Faizan-Alam-1, 2024-05-03:
Can you please let me know exactly what I need to improve? I have noticed that the code contains many comments and repeated lines of code. Do I need to improve that?

This f1 model code.
https://github.com/JdeRobot/RoboticsInfrastructure/blob/humble-devel/CustomRobots/f1/models/f1/model.sdf

pariaspe, 2024-05-03:
Hi @Faizan-Alam-1, current SDF models have a lot of duplicated code. The idea is to create a base model (f1_base for instance) and include that base model in the final ones. You can take a look at the [SDFormat specification](http://sdformat.org/spec?ver=1.9&elem=sdf) docs.

Faizan-Alam-1, 2024-05-04:
For further clarification, I'm providing an example. Please review it. Is this the way you talk about it?

## First, I will create a file named f1_base.sdf with the following code inside:

```
<link name="f1_base">
  <include>
    <uri>#f1</uri>  
  </include>
 <inertial>
    <mass>1</mass>
    <inertia>
      <ixx>0.5</ixx>
      <ixy>0.0</ixy>
      <iyy>0.5</iyy>
      <ixz>0.0</ixz>
      <iyz>0.0</iyz>
      <izz>1.0</izz>
    </inertia>
  </inertial>
  <collision name="collision">
    <geometry>
      <cylinder>
        <radius>0.06</radius>
        <length>0.07</length>
      </cylinder>
    </geometry>
  </collision>
</link>

<link name="front_right_wheel">
  <pose>0 -0.16 0.06 0 1.5707 0</pose>
  <include>
    <uri>#f1_base</uri>
  </include>
</link>

<link name="front_left_wheel">
  <pose>0 0.16 0.06 0 1.5707 0</pose>
  <include>
    <uri>#f1_base</uri>
  </include>
</link>

```
For example, I'm including only front_right_wheel and front_left_wheel, but I will add all components of the f1 model.

## In the existing f1 model, the .sdf file will contain this code:

```
<?xml version='1.0'?>
<sdf version="1.4">
<model name="f1">
  <pose>0 0 0 0 0 0</pose>
  <static>false</static>
    <link name="f1">
    	<pose>0 0 0 0 0 0</pose>
	<inertial>
        <mass>10</mass>
        <inertia>
          <ixx>1</ixx>
          <ixy>0.0</ixy>
          <iyy>1</iyy>
          <ixz>0.0</ixz>
          <iyz>0.0</iyz>
          <izz>1.0</izz>
      </inertia>
    </inertial>

      <collision name="collision">
            <geometry>
              <mesh>
            		<uri>model://f1/meshes/F1.dae</uri>
          		</mesh>
          </geometry>
     </collision>
      <visual name="visual">
        <geometry>
          <mesh>
            <uri>model://f1/meshes/F1.dae</uri>
        </mesh>
      </geometry>
    </visual>
  </link>

```

Faizan-Alam-1, 2024-05-04:
@pariaspe  Please let me know if this is the way you want it. If not, please provide some more explanation.

pariaspe, 2024-05-04:
> @pariaspe Please let me know if this is the way you want it. If not, please provide some more explanation.

Kind of, what we need is a base model (`f1_base`) and several extensions of it: `f1_cam`, `f1_laser`, etc

Take a look at this example:
```xml
<?xml version="1.0" ?>
<sdf version="1.6">
<model name="f1_renault_cam">
  <pose>0 0 0 0 0 0</pose>
  <static>false</static>

    <include merge="true">
        <uri>model://f1</uri>
        <name>f1</name>
    </include>

    <link name='camera_link'>
         ............
    </link>

    <joint type="fixed" name="camera_joint">
      <parent>f1::f1_link</parent>
      <child>camera_link</child>
    </joint>

  </model>
</sdf>
```

Bsprogrammer28, 2024-10-03:
Is this issue Still open?

pariaspe, 2024-10-08:
Not for noetic anymore. But you can take a look at humble models and check if models are nested

closed 2024-10-08 by pariaspe