JdeRobot/RoboticsInfrastructure #221: Question: Can jderobots/drones assets be moved here?

damodardatta, 2023-04-16:
@jmplaza , wanted to ask, as you mentioned in [here](https://github.com/JdeRobot/BehaviorMetrics/issues/601#issuecomment-1501125824) about [CustomRobots repository](https://github.com/JdeRobot/CustomRobots) being refactored into RoboticsInfrastructure repository. Wouldn't it make more sense to move drone assets from [jderobot/drones](https://github.com/JdeRobot/drones/tree/noetic-devel) to here? And letting drones repo build upon this. 

@pariaspe want your opinion as well. As do this can make upgrading drones to ROS 2 easier.

pariaspe, 2023-04-17:
Hey @damodardatta , drones rely on PX4 which is a bit unstable this days with ROS 2. We'll wait some time to see the evolution of PX4 project before moving things. 
Moreover, we probably drop PX4 in favor of pure Gazebo simulation as soon as we move to new Gazebo (former Ignition Gazebo).

damodardatta, 2023-04-19:
@pariaspe  Do you mean something like [this](https://github.com/fdcl-gwu/uav_simulator)?

pariaspe, 2023-04-19:
Yes, something similar but using new Gazebo. More like [this](https://aerostack2.github.io/_02_examples/python_swarm_drones/index.html).

pariaspe, 2023-06-19:
Closed in #297

closed 2023-06-19 by pariaspe