JdeRobot/RoboticsInfrastructure #240: [Humble] Webserver not running in Humble RADI by default

pawanw17, 2023-05-09:
Hi I am trying to test the Humble RADI, but the webserver doesn't seem to be running with the entrypoint.sh script. 

To Reproduce:

I built the RADI using the build.sh script https://github.com/JdeRobot/RoboticsInfraestructure/blob/humble-devel/build.sh

and ran it using

`./run.sh -t humble_8_may`

Here are the RADI logs:

> pawan@pawan-Legion:~/devel/RoboticsInfraestructure/scripts$ ./run.sh -t humble_8_may 
name not specified, container name will be: dockerRam_container
environment set
INFO:root:Starting RAM consumer in <websocket_server.websocket_server.WebsocketServer object at 0x7f1294ea8160>:7163
2023-05-09 16:37:07,080 [MainThread  ] [INFO ] (root)  Starting RAM consumer in <websocket_server.websocket_server.WebsocketServer object at 0x7f1294ea8160>:7163
INFO:websocket_server.websocket_server:Listening on port 7163 for clients..
2023-05-09 16:37:07,081 [MainThread  ] [INFO ] (websocket_server.websocket_server)  Listening on port 7163 for clients..
INFO:websocket_server.websocket_server:Starting WebsocketServer on thread Thread-1 (serve_forever).
2023-05-09 16:37:07,081 [MainThread  ] [INFO ] (websocket_server.websocket_server)  Starting WebsocketServer on thread Thread-1 (serve_forever).

[image]

jmplaza, 2023-05-09:
Hi @pawanw17 , you are in the cutting cutting edge :-). 

We have recently reorganized the repos and now we don't have a copy of RoboticsAcademy and a copy of RAM in this "CustomRobots" repo. Instead, the RAM has been moved to a [standalone repo](https://github.com/JdeRobot/RoboticsApplicationManager) and CustomRobots repository has been renamed as RoboticsInfrastructure, including now a CustomRobots _directory_. So currently ROS2-Humble based flavour of RoboticsAcademy is broken. Let's wait some days until we put it up & running again with the brand new repo organization.

pawanw17, 2023-05-09:
Thanks for the update @jmplaza! I'll pick up something else to do while restructuring finishes, let me know if I can help.

pawanw17, 2023-05-10:
Thanks @Blancasr I am able to run it now! 
[image]

jmplaza, 2023-05-18:
Hi @pawanw17 , could you please test it with the brand new RADI 4.4.1?

pawanw17, 2023-05-18:
Hi @jmplaza The webserver is running as expected in RADI 4.4.1. Closing the issue.

closed 2023-05-18 by pawanw17