JdeRobot/RoboticsInfrastructure #258: [Humble] Missing manager.py in RADI 4.4.0

pawanw17, 2023-05-17:
Hi, during testing the recently pushed RADI 4.4.0 I am facing this particular issue:

[image]

with -d flag and running entrypoint.sh the exercises do not come to ready state.
[image]

[image]

ReyDoran, 2023-05-18:
Hi, sorry I think it's due to a mistake I made on entrypoint.sh while uploading the image. Could you please retry downloading the RADI 4.4.0?

pawanw17, 2023-05-18:
Hi @ReyDoran Fixed. Thanks!

closed 2023-05-18 by pawanw17