JdeRobot/RoboticsInfrastructure #337: [Humble] Duplicate worlds in the repository

pawanw17, 2023-10-11:
There exist duplicate worlds inside Worlds directory and CustomRobots/

- roomba_robot/worlds
- ackermann_cars/worlds
- car_junction/worlds
- amazon_robot/worlds
- amazon_hospital/worlds
- autoparking/worlds

Tanishq30052002, 2024-03-05:
Hi @pawanw17 ,
I would like to work on the issue, can i go ahead with it?

pawanw17, 2024-03-12:
Sure please

Tanishq30052002, 2024-03-13:
@pawanw17 could you please tell me on which branch I am supposed to work?
Since the branch used in the humble-4.4.32 is an issue branch and has a different directory structure.

Tanishq30052002, 2024-03-13:
@pawanw17 should I just remove the same world files and move the different world files to the WORLD directory?
since there are some world world files which are not present in the world directory but yet also not being used in any launch file also.

Tanishq30052002, 2024-03-13:
hey @pawanw17 ,
In the repo there are different worlds files and different launch files too, so am I supposed to convert them to launch.py format too so it will be compatible with ROS2 also. Also, many more pkgs have the same problem, so could you please clarify it?

pawanw17, 2024-03-16:
Hi @Tanishq30052002, In this repo mainly we have two branches which are actively developed, 
noetic-devel and humble-devel, inside these you can find .world files which gazebo uses, This issue is only targetting the world files and not the launch files.

Tanishq30052002, 2024-03-16:
Should I add different world files too in world directory?

pawanw17, 2024-03-16:
What do you mean?

Tanishq30052002, 2024-03-16:
@pawanw17 
Since in custom robots there are some more world files which are not duplicate to the ones in the world directory.
So I want to know if we are supposed to move those world files to the world directory also?

pawanw17, 2024-03-19:
@jmplaza where should we keep all the world files? inside CustomRobots folder?

That way when CustomRobots is built they would automatically be sourced, but I don't know if other applications(Visualcircuit, btstudio) build CustomRobots.

Tanishq30052002, 2024-03-28:
hi @pawanw17 @jmplaza 
I would be glad if you could please clear the task.